Protocentral/healthypi-move-fw #39: west build issue ... cannot find custom board

Hoot-Vaughn, 2025-03-18:
I am able to start a virtual enviroment and run 

west init -m https://github.com/protocentral/healthypi-move-fw --mr main healthypi-move-workspace
cd healthypi-move-workspace
west update

however, when i try to build the firmware using 

west build -b healthypi_move_nrf5340_cpuapp application/app
west flash

I get an error saying that the path "application/app" is not valid see below:

...healthypi-move-workspace % west build -b healthypi_move_nrf5340_cpuapp application/app
west flash
ERROR: source directory application/app does not exist
FATAL ERROR: refusing to proceed without --force due to above error
FATAL ERROR: --build-dir was not given, and neither /Users/joshvaughn/Documents/ChemoCodeNEW2/healthypi-move-workspace nor /Users/joshvaughn/Documents/ChemoCodeNEW2/healthypi-move-workspace/build are zephyr build directories.

I tried editing the path to be simply "app" as I see no "application" in the cloned files. However when I run the code:

west build -b healthypi_move_nrf5340_cpuapp app            
west flash

    The build tries to initate, but then gives this error:
    No board named 'healthypi_move_nrf5340_cpuapp' found.
    
    Please choose one of the following boards:
    
    healthypi_move
    thingy91
    nrf52kbd
    thingy91x
    nrf52833dongle
    nrf52810dmouse
    nrf52840gmouse
    nrf52820dongle
    nrf52dmouse
    samr34_xpro
    samc21n_xpro
    same54_xpro
    saml21_xpro
    samd21_xpro
    samd20_xpro
    samr21_xpro
    sam_e70_xplained
    sam_v71_xult
    sam4l_ek
    sam4s_xplained
    sam4e_xpro
    wio_terminal
    lora_e5_mini
    xiao_esp32c3
    lora_e5_dev_board
    xiao_ble
    xiao_rp2040
    xiao_esp32s3
    seeeduino_xiao
    mercury_xu
    cc26x2r1_launchxl
    cc3220sf_launchxl
    cc1352p1_launchxl
    cc1352p7_lp
    msp_exp432p401r_launchxl
    sk_am62
    cc3235sf_launchxl
    cc1352r1_launchxl
    cc1352r_sensortag
    robokit1
    acrn
[... 565 lines cut ...]
    nrf21540dk
    nrf51dongle
    nrf9280pdk
    nrf7002dk
    intel_ehl_crb
    intel_ish_5_4_1
    intel_ish_5_6_0
    intel_ish_5_8_0
    niosv_m
    intel_adl_crb
    intel_adl_rvp
    niosv_g
    intel_adsp
    intel_rpl_p_crb
    intel_rpl_s_crb
    cyclonev_socdk
    intel_socfpga_agilex_socdk
    intel_socfpga_agilex5_socdk
    decawave_dwm1001_dev
    roc_rk3568_pc
    esp32c3_042_oled
    CMake Error at /Users/.../Documents/ChemoCodeNEW2/healthypi-move-workspace/external/zephyr/cmake/modules/boards.cmake:228 (message):
      Invalid BOARD; see above.

I have tried google searching but have not found anything of help so far.... any ideas?

protocentralashwin, 2025-03-22:
The west build command using the new hardware model terms for the board, so for referring to the the cpuapp core it would be  healthypi_move/nrf5340/cpuapp. So the west build command becomes:

`west build -b healthypi_move_nrf5340_cpuapp app`

I hope this helps

Hoot-Vaughn, 2025-03-22:
I tried that and the command goes to build the board, but then gives an error that it cannot find it. I am not sure if I did something wrong with the cloning process? but I create a new file path then created a virtual environment and downloaded west. Then I used the following code to clone the repositories in a new workspace

    west init -m https://github.com/protocentral/healthypi-move-fw --mr main healthypi-move-workspace
    cd healthypi-move-workspace
    west update

after that I ran 

    west build -b healthypi_move_nrf5340_cpuapp app

It starts to build but then cannot find the board.... the exact error I get is copied below
Also I am running this in the terminal on VScode (not sure if that is relevant or not)

        (.venv) (base) baseuser@Base-MBP-3 healthypi-move-workspace % west build -b healthypi_move_nrf5340_cpuapp app
        -- west build: generating a build system
        Loading Zephyr module(s) (Zephyr base): sysbuild_default
        -- Found Python3: /Users/baseuser/Documents/SouceCode/watch/.venv/bin/python3.13 (found suitable version "3.13.2", minimum required is "3.8") found components: Interpreter
        -- Cache files will be written to: /Users/baseuser/Library/Caches/zephyr
        -- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
        No board named 'healthypi_move_nrf5340_cpuapp' found.
        
        Please choose one of the following boards:
        
        healthypi_move
        thingy91
        nrf52kbd
        thingy91x
        nrf52833dongle
        nrf52810dmouse
        nrf52840gmouse
        nrf52820dongle
        nrf52dmouse
        samr34_xpro
        samc21n_xpro
        same54_xpro
        saml21_xpro
        samd21_xpro
        samd20_xpro
        samr21_xpro
        sam_e70_xplained
        sam_v71_xult
        sam4l_ek
        sam4s_xplained
        sam4e_xpro
        wio_terminal
        lora_e5_mini
        xiao_esp32c3
        lora_e5_dev_board
        xiao_ble
        xiao_rp2040
        xiao_esp32s3
        seeeduino_xiao
        mercury_xu
        cc26x2r1_launchxl
        cc3220sf_launchxl
        cc1352p1_launchxl
        cc1352p7_lp
        msp_exp432p401r_launchxl
        sk_am62
        cc3235sf_launchxl
        cc1352r1_launchxl
        cc1352r_sensortag
        robokit1
        acrn
        acrn_ehl_crb
        gr716a_mini
        generic_leon3
        rpi_5
        rpi_4b
        rpi_pico
[... 568 lines cut ...]
        intel_adl_crb
        intel_adl_rvp
        niosv_g
        intel_adsp
        intel_rpl_p_crb
        intel_rpl_s_crb
        cyclonev_socdk
        intel_socfpga_agilex_socdk
        intel_socfpga_agilex5_socdk
        decawave_dwm1001_dev
        roc_rk3568_pc
        esp32c3_042_oled
        CMake Error at /Users/baseuser/Documents/SouceCode/healthypi-move-workspace/external/zephyr/cmake/modules/boards.cmake:228 (message):
          Invalid BOARD; see above.
        Call Stack (most recent call first):
          cmake/modules/sysbuild_default.cmake:15 (include)
          /Users/baseuser/Documents/SouceCode/healthypi-move-workspace/external/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
          /Users/baseuser/Documents/SouceCode/healthypi-move-workspace/external/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
          /Users/baseuser/Documents/SouceCode/healthypi-move-workspace/external/zephyr/share/sysbuild-package/cmake/SysbuildConfig.cmake:8 (include)
          template/CMakeLists.txt:10 (find_package)
        
        
        -- Configuring incomplete, errors occurred!
        FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake -DWEST_PYTHON=/Users/baseuser/Documents/SouceCode/watch/.venv/bin/python3.13 -B/Users/baseuser/Documents/SouceCode/healthypi-move-workspace/build -GNinja -DBOARD=healthypi_move_nrf5340_cpuapp -S/Users/baseuser/Documents/SouceCode/healthypi-move-workspace/external/zephyr/share/sysbuild -DAPP_DIR:PATH=/Users/baseuser/Documents/SouceCode/healthypi-move-workspace/app
        (.venv) (base) baseuser@Base-MBP-3 healthypi-move-workspace %

protocentralashwin, 2025-03-26:
I'm sorry, what I meant was, the west build command becomes this:

`
west build -b healthypi_move/nrf5340/cpuapp app
`

Hoot-Vaughn, 2025-03-27:
Ok that fixed the problem for this issue I now have another but I will open another thread for it. Thanks

closed 2025-03-27 by Hoot-Vaughn