Protocentral/healthypi-move-fw #44: Unknown CMake command "add_subdirectory_ifdef"

Hoot-Vaughn, 2025-03-27:
during the build process after I am getting a error that causes the failure of the build:

    CMake Error at drivers/CMakeLists.txt:3 (add_subdirectory_ifdef):
      Unknown CMake command "add_subdirectory_ifdef".
    -- Configuring incomplete, errors occurred!
    CMake Error at cmake/modules/sysbuild_extensions.cmake:514 (message):
      CMake configure failed for Zephyr project: app
  
not sure what is causing this but just looking up google looks like the add_subdirectory_ifdef isnt a CMake predefined function? maybe it has to do with my version of CMake? I am currently running:

    cmake version 3.31.6

Thank you for the help.

uutzinger, 2025-04-10:
I used git to clone the repo, then installed the nRF Code Extension that includes nRF Conenct, Terminal, Device Tree, Kconfig, C/C++, Linker, and Cmake. In VS Code nRF extension I selected 2.9.0 tool chain. I also installed Segger JLink (likely not needed) and nrf-command-line-tools with separate download. Then in VS Code I opened existing application in the cloned git folder and added the board config as shown on the website move.protocentral.com "for developers" section. I was able to build the app and it created dfu_applicatioin.zip in the app/build folder. This file can be uploaded to the watch without the nRF 5430 dk (the board shown on move website and not included with the watch) using your phone  after installing nRF Connect (Mobile App) or on Windows with the nRF Connect Desktop by connecting to the watch with app. It scans for all bluetooth devices and is quite helpful. The watch should be connected to USB C power for optimal firmware update. In the nRF Connect application on your phone or Windows you can hit DFU symbol and upload the zip file. You can also obtain the firmware zip file on this repo under assets.

Hoot-Vaughn, 2025-04-15:
Ok so I have followed what you said above, however, when I "open existing application" via 
healthypi-move-workspace\app\app. 

The application opens and shows that I am using "nrf connect SDK v2.9.0" and "nRF Connect SDK Toolchain v2.9.0". However the "Board Target" drop down shows "No boards found". 

I am using a Mac so I am not sure if that has anything to do with it. I am currently also trying to do the same thing on dual boot using Ubuntu. But I don't hope for it to help. Any ideas?

Hoot-Vaughn, 2025-04-15:
just finished the same exact process in ubuntu. I did still use west to clone both repositories on the MacOS and Ubuntu.

protocentralashwin, 2025-04-23:
@Hoot-Vaughn are you still facing this issue?

Hoot-Vaughn, 2025-04-23:
still having this issue. I am not sure what to do. I tried reaching out directly to http://protocentral.com customer support, but have not received anything. The nrf sdk tool chain seems really buggy. but I have tried doing this both manually from command line and from VS Code with no luck.

uutzinger, 2025-04-24:
I have many boards to choose from as you can see below. I use Ubuntu based system.
Just clone the repository with git. I did not use west to download.

[image]