Protocentral/healthypi-move-fw #49: Move is not restating after updating with v0.9.38

shagray, 2025-04-23:
My move is not restating after updating with version v0.9.38

Kindly help with some solution

protocentralashwin, 2025-04-23:
Did the update upload fully? Which app did you use for the update?

You can do a forced reset by keeping the side button pressed for 10 secs, and leave it when it restarts.

shagray, 2025-04-23:
Here is what I have done 

I initially upgraded to v0.9.37 using nrf connect app, it was working fine

Then I upgraded to v1.0.0 and noted that blood pressure feature is not available, which was present in the v0.9.37

So I decided to downgrade to v0.9.38 from v1.0.0.

nrf connect showed success message, but move didn’t restart 

Tried forced rest,  but no luck,

Whether my move is bricked?

protocentralashwin, 2025-04-23:
Thank you for the details. 

We do not support BPT yet, and the one in 38 was not fully complete. We haven't explicitly disabled downgrading, but it should work. Is the device BLE advertising?

Can you charge for sometime and try to update again while USB is connected?

shagray, 2025-04-23:
BLE is not advertising 

There is no indicator such as green/red at the bottom portion of move

During the upgrade/ downgrade , battery was around 80%

Charging it now ,will update later

stardust-an, 2025-04-23:
Dear Author,

Hello! I encountered an issue while performing a DFU update following the instructions in the "Update using nRF Connect" section of the documentation. I was able to proceed normally up to the step: "Select Confirm only in the dialogue box appears, click on Ok and the firmware updating will begin." However, during the firmware transfer, the progress bar reaches about halfway, and then the watch becomes unresponsive. At that point, I can no longer perform any operations and have to force a shutdown and restart the device.

I have tried several different versions of the update package, but the result has been the same each time. Could you please help me understand what might be causing this issue? I would greatly appreciate your support.

For reference, the hardware I am using is the HEALTHPI-MOVE-BASIC, and the current firmware version is V0.9.14.

Thank you very much in advance!

[Log 2025-04-23 19_46_02.txt](https://github.com/user-attachments/files/19866638/Log.2025-04-23.19_46_02.txt)

protocentralashwin, 2025-04-23:
Hellob @stardust-an , 

can you try the same process once more with nrf device manager? It's available on the app store and play store. Sometimes nrfconnect has issues. 

Which os are you using?

protocentralashwin, 2025-04-24:
@stardust-an , have you tried to completely reset the device (by pressing the button for 10 secs) and then try this process again?

Also, I see that you're trying to upload 0.9.35, can you use 1.0.0 only?

Hopefully all these issues will go away once our app's DFU capabilities are ready

stardust-an, 2025-04-24:
Hello again,

I just attempted to upload the v1.0.0 update package. I also tried your suggestion of restarting the device and attempting the update again, but unfortunately, the issue still persists. The device either becomes unresponsive or enters a reboot loop, and eventually the app displays an error message.

In addition, I noticed there are some optional settings under the Firmware Upgrade section in the app. Could it be possible that one of these default settings is affecting the DFU process?

Lastly, is there any additional information I can provide to help troubleshoot this issue? I would really appreciate any guidance on how to resolve it.

[image]

protocentralashwin, 2025-04-25:
@shagray I'm sorry you're still having this issue. No, usually these settings do not matter since the mcuboot bootloader on the device has it's own settings. 

Do you have anything other Android or iOS phone with you? if so, can you try one more time with that? 

In case nothing works, do you have a JTAG debugger on you, or an nrf development board?