Protocentral/healthypi-move-fw #51: What is the watch able to measure and store with V 1.0.0 firmware and the Move app?

tony1tf, 2025-04-24:
The watch has been working for a day. Why does the app only display heart rate - not SpO2, Temp or Activity.  When I sync it, it fetches files OK. Also, can it show temp in deg C?

protocentralashwin, 2025-04-25:
It is supposed to fetch and display and all the files. Let me check with our app developer. Just to confirm, which version of the app do you have?

Based on multiple user suggestions, we are adding the temperature in deg C also in the coming release.

tony1tf, 2025-04-25:
The app is the early access version 0.2.0. of Move. I've just sent some feedback to the Play Store.

protocentralashwin, 2025-04-25:
You should have access to v1.0.0 in the open testing channel. Now it shows that it has been released. Maybe google had some delay in releasing.

tony1tf, 2025-04-25:
Thanks for the nudge. I've just updated to 1.0.0. It doesn't make any difference, apart from orange range sliders to zoom in on data. Still only BP, nothing else, and nothing from today yet. Another issue, the watch battery seems to discharge quickly - in c. 2 hrs it's down from 100% to 80%. It is dead by early evening.

shagray, 2025-04-27:
App don’t display SpO2 and Activity. It displays only heart rate and temperature.

protocentralashwin, 2025-04-27:
@shagray @tony1tf , yes there is a problem with the activity recording due to a bug in imu step counter code. I will release a new update shortly fixing this issue only.