Protocentral/healthypi-move-fw #54: Unable to Connect HealthyPi-move to J-Link Debugger for Flashing New Firmware

PingAnanan, 2025-04-27:
Dear Author,

I am reaching out for assistance regarding an issue I encountered while attempting to update the firmware on my HealthyPi-move device. After facing problems with DFU as discussed in [this issue](https://github.com/Protocentral/healthypi-move-fw/issues/49), I followed your suggestion to flash a different version of the firmware.

However, during the process, I ran into a new issue: my HealthyPi-move is not connecting to the J-Link debugger, which prevents me from flashing the firmware. Here's how I discovered the problem:

When I used the nRF5340-DK development board, I noticed that the flash tool would display the chip model of the target device during the flashing process. But when I connected my HealthyPi-move to the J-Link as per the documentation, the chip model did not appear, indicating that the debugger couldn't establish a connection.

I tried both the external interface of the 5340-DK's onboard programmer and a separate standalone J-Link, but the result was the same.

Could you please advise on what might be causing this issue? Any suggestions on how to resolve it would be greatly appreciated.

Thank you for your help!

[image]

PingAnanan, 2025-04-27:
I confirm that my hardware connections were done exactly according to the steps in the documentation.

PingAnanan, 2025-04-27:
To troubleshoot further, I disassembled the HealthyPi-move and manually soldered fly wires to the SWD debug interface, connecting it directly to the J-Link.

After this, I successfully flashed the v1.0.0 firmware to the HealthyPi-move.
However, the DFU problem still persists: when I try to perform a DFU update, the upload gets stuck halfway, exactly like before.

protocentralashwin, 2025-04-27:
@PingAnanan looks you've done quite a bit of work now !! 

You mean the jtag adapter board, through USB is not working? Did you try reversing the USB c connector? Sometimes small particles on the pins can cause this. 

Regarding the DFU, I'm still not sure why, but once we remove the dependency on Nordic apps, it should be fine I hope. 

We would love any feedback you have, good or bad about the overall experience,  iv3n that you've gone upto the soldering level !

protocentralashwin, 2025-04-27:
Also, if you think something might be wrong with the jtag adapter board, even after you've reversed the USB connector, please let us know and we can send you a replacement.

protocentralashwin, 2025-07-09:
I am closing this issue for now. If you still face any related issue, please feel free to create a new issue.