Protocentral/healthypi-move-fw #58: debugging won't-boot situation

jjfiv, 2025-05-04:
Hey y'all,

My healthypi move worked out of the box. I was having a similar issue (to https://github.com/Protocentral/healthypi-move-fw/issues/49) flashing mine with the generic nRF apps (I'm on iOS so I think that's still the recommended path). The upload would crash out halfway through and the watch would hang; I hard reset it a few times and kept trying, and then eventually just let it run out of battery to wait for the real app/official 1.0 firmware.

Now I have black screen; no booting, etc. I was able to hook up the whole debug/flash setup as described in the instructions, and I've tried flashing the release files as follows:

```
$ nrfutil device program --firmware healthypi_move_v1.0.5.hex # downloaded from github releases
[00:00:07] ###### 100% [2/2 1050047064] Programmed
```

The board seems responsive over JTAG (as evidenced by my ability to program "successfully".

But I'm still not able to get a sign of life from the healthypi move (no blinking LEDs, no screen). I don't believe I've used an inappropriate USB-C cable or anything with the device; and gently disassembling the pieces, I do not see any burned out components.

Hoping I managed to get it to a bad state flashing and if you point me to what files I should be using it will come back to life 🤞.

Let me know if there's any more debugging information I can collect!

protocentralashwin, 2025-05-05:
Thanks @jjfiv for providing detailed information and I'm sorry the DFU didn't work out. 

It's good that the display was working when you took it out, suggesting that the display is not damaged.

Regarding the programming, can you try using the older nrfjprog tools instead of the nrfutil? (https://docs.nordicsemi.com/bundle/ug_nrf_cltools/page/UG/cltools/nrf_command_line_tools_lpage.html). We've seen issues sometimes with nrfutil and nRF5340. 

You can also try compiling from source and flashing directly to see if that works.

jjfiv, 2025-05-06:
Thanks @protocentralashwin! My healthypi indeed came back to life; I did end up flashing from (probably that button runs west flash), don't know if there was a shorter route on the CLI (needed to "recover" before I could program). The older toolchain took a _much_ longer time to clear nvram, which was a good sign.

In hindsight, I should have definitely paid more attention to the versions used in the tutorial screenshots; don't know why I fell for the "The old one's deprecated? I'll just try using the latest version of an embedded toolchain..." rookie mistake 🤣

closed 2025-05-06 by jjfiv