Protocentral/healthypi-move-fw #62: Firmware V1.1.0

tony1tf, 2025-05-24:
I noticed that you have just issued a new firmware release. I am happy to report successful upload to the watch. It updated the max32664 or d, and went straight to an SPO2 measurement. As yet, I do not see an ECG measurement - is this coming soon?
Could you also explain why the current heart rate reading doesn't appear in the history graphs on the app. 
Tony

uutzinger, 2025-05-24:
I have access to all pannels in V1.1.0 including blood pressure and ecg and temperature

protocentralashwin, 2025-05-25:
@tony1tf @uutzinger 1.1.0 still has some issues during our testing, which is why we marked it as pre-release. We will make a follow-up release in a couple of days, along with the accompanying app release.