Protocentral/healthypi-move-fw #68: V1.2 Flashing using nRF5340 DK

daniel-edwards16, 2025-05-31:
Hi,

I'm attempting to flash the new v1.2 onto the device through VSCode. I have followed all the steps in [here](https://move.protocentral.com/development/03-flashing_and_debugging/), but when flashing it fails to verify. My watch no longer boots. Here is the full terminal output while flashing.

`Executing task: nRF Connect: Flash: app/build_2 (active) 

Flashing build_2 to 1050045939
C:\WINDOWS\system32\cmd.exe /d /s /c "west flash -d c:\Users\danst\move\healthypi-move-workspace\app\app\build_2 --dev-id 1050045939 --recover"

-- west flash: rebuilding
[0/26] Performing build step for 'ipc_radio'
ninja: no work to do.
[1/26] Performing build step for 'app'
ninja: no work to do.
[4/26] cmd.exe /C "cd /D C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\_sysbuild && C:\ncs\toolchains\b620d30767\opt\bin\cmake.exe -E true"
[5/22] Performing build step for 'mcuboot'
ninja: no work to do.
[8/22] cmd.exe /C "cd /D C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\_sysbuild && C:\ncs\toolchains\b620d30767\opt\bin\cmake.exe -E true"
[9/19] Performing build step for 'b0n'
ninja: no work to do.
[12/19] cmd.exe /C "cd /D C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\_sysbuild && C:\ncs\toolchains\b620d30767\opt\bin\cmake.exe -E true"
[14/18] No install step for 'b0n'
[15/18] Completed 'b0n'
[16/18] cmd.exe /C "cd /D C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\_sysbuild && C:\ncs\toolchains\b620d30767\opt\bin\cmake.exe -E true"
WARNING: Specifying runner options for multiple domains is experimental.
If problems are experienced, please specify a single domain using '--domain <domain>'
-- west flash: using runner nrfjprog
-- runners.nrfjprog: reset after flashing requested
-- runners.nrfjprog: Recovering and erasing flash memory for both the network and application cores.
Recovering device. This operation might take 30s.
Erasing user code and UICR flash areas.
Writing image to disable ap protect.
Recovering device. This operation might take 30s.
Erasing user code and UICR flash areas.
Writing image to disable ap protect.
-- runners.nrfjprog: Flashing file: C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\merged_CPUNET.hex
[ #################### ]  12.222s | Erase file - Done erasing                                                          
[ #################### ]   1.365s | Program file - Done programming                                                    
[ #################### ]   1.328s | Verify file - Done verifying                                                       
Applying system reset.
Run.
-- runners.nrfjprog: Board with serial number 1050045939 flashed successfully.
-- west flash: using runner nrfjprog
-- runners.nrfjprog: reset after flashing requested
-- runners.nrfjprog: Flashing file: C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\merged.hex
[ #################### ]   0.000s | Erasing non-volatile memory - Erase successful                                     
[ #################### ]  34.098s | Erase file - Done erasing                                                          
[ #################### ]  13.126s | Program file - Done programming                                                    
[error] [ Client] - Encountered error -160: Command verify_file executed for 10772 milliseconds with result -160       
[error] [  nRF53] - Failed while performing 'Verify' operation on target address 0x10120000. e                         
-160: Data does not match in address range [0x10120000 - 0x101949D6] (XIP (QSPI))
Expected byte value 0x3D but read 0x00 at address 0x10120000.
[error] [  nRF53] - Failed while verifying device. -160: Data does not match in address range [0x10120000 - 0x101949D6] (XIP (QSPI))
Expected byte value 0x3D but read 0x00 at address 0x10120000.
[error] [ Worker] - Data does not match in address range [0x10120000 - 0x101949D6] (XIP (QSPI))
Expected byte value 0x3D but read 0x00 at address 0x10120000.
ERROR: Write verify failed.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
FATAL ERROR: command exited with status 25: nrfjprog --program 'C:\Users\danst\move\healthypi-move-workspace\app\app\build_2\merged.hex' --sectorerase --qspisectorerase --verify --qspiini C:/Users/danst/move/healthypi-move-workspace/app/app/qspi_w25.ini -f NRF53 --coprocessor CP_APPLICATION --snr 1050045939`

protocentralashwin, 2025-05-31:
Thanks @daniel-edwards16 for reporting this.

This could happen if your previous version was before 1.0.x, since the nrf partition manager sometimes is unable to locate the correct address for XIP.

Can you please run the following command from a terminal where nrfnprog is accessible to delete the QSPI flash completely?

`nrfjprog  --qspieraseall`

After that, please try flashing using the same method that you used earlier.

daniel-edwards16, 2025-05-31:
using `west flash --recover` results in the same error message as my original post. I am not sure how to check if its programming the watch or the DK. I've attached some images for more info.

[image]

[image]

daniel-edwards16, 2025-05-31:
Upon further inspection, I believe that it is trying to flash to my DK board (1050045939)