Protocentral/healthypi-move-fw #69: Firmware v1.20

tony1tf, 2025-06-02:
Uploaded OK, along with Android app autoupdate.
Will show live ppg, but not sync and download heart rate data. Orange bar just keeps sweeping until I hit cancel.
Tony

protocentralashwin, 2025-06-11:
Hi @tony1tf, have you tried with the latest app version (v1.2.2)? 

Also, after updating, please delete all app local data using the button in the "Settings" tab and then retry syncing?

tony1tf, 2025-06-12:
Hi Ashwin

The latest firmware on github is v1.2.0. Where is V1.2.2?

Tony

awhitchurch, 2025-10-25:
Please check with FW v1.9.0 and feel free to open a new issue: https://www.crowdsupply.com/protocentral/healthypi-move/updates/major-improvements-new-firmware-and-new-enclosure

closed 2025-10-25 by awhitchurch