Protocentral/healthypi-move-fw #77: MAX32664C PPG Sensor Reading Issue

daniel-edwards16, 2025-07-04:
Hi again,

I'm still developing my custom application based on the source firmware and hardware provided and I wanted to ask about lines 180-310 (function max32664c_async_sample_fetch) in max32664c_async (linked below)

https://github.com/Protocentral/healthypi-move-fw/blob/3de9c1a482fb429bcc9f8aa86d71a61703a43d03/drivers/sensor/max32664c/max32664c_async.c#L180C1-L310C2

I am struggling to understand why sample_len = 48 when in normal mode and sample_len = 70 when in extended mode. Table 8 in the [MAX32664C datasheet](https://www.analog.com/media/en/technical-documentation/user-guides/measuring-spo2-and-heart-rate-using-max32664c.pdf) shows that the normal algorithm returns a total of 44 bytes and the extended algorithm returns a total of 76 bytes. I'm having issues with reading the raw red and IR led values returned from this function.

Photos of the datasheets below:

[image]

[image]

Any help would be much appreciated, also please note I am fairly new to the zephyr and embedded programming processes.

Thanks in advance,
Daniel

daniel-edwards16, 2025-07-04:
So quick update, I reverted my modified code back to what was originally there and now it works? I am not sure why, but I am not going to touch it again. I'll leave this post up for anyone else if they get confused too

protocentralashwin, 2025-07-09:
hello @daniel-edwards16 : I'm sorry for the delayed reply. I think it's because you're looking at the data sheet for the older sensor hub firmware. For the latest document, you can download the datasheet after you register and download the file from here: https://www.analog.com/en/products/max32664.html

The sensor hub on the Move runs the FW version 30.13.31